slic3r/Slic3r-Manual #29: question about bed shape (size and origin)

ruggb, 2015-10-12:
I have 2 build areas and 2 hotends
the total build size is 305x305mm without hot bed.
the total X movement is 335mm to accommodate hotend 2, which is located -30, 0 from hotend 1
The hotbed build size is 201x194 placed with origin at 50, 70
for non-hotbed use is the bed size = 305x305 or 335x305 ? Marlin max is 335x305
for hotbed use I have no idea what numbers to use.
can u clear this up for me??

lordofhyphens, 2017-03-22:
This isn't the appropriate place for this. 
#reprap on freenode irc is a good place to ask this kind of question.

Use whatever works for you, the bed size is just there to aid you in producing correct gcode for your printer.

ruggb, 2017-03-22:
hey, sh, if it is "there to aid you in producing correct gcode for your printer, and Slic3r produced gcode, the I think your answer is VERY inappropriate.
just saying.

lordofhyphens, 2017-03-22:
This is the bug/issue tracker for the manual (incorrect content, errors, etc) Your question was inappropriate. I pointed you at a better place to get help and noted that the actual bed size is not particularly important. 

Now, to answer it as best as possible:

Use the size of printbed as fits your physical printing area in either case. Use two profiles, one with heatbed and one without. 

You may need to adjust the origin point (or use m206) to get both profiles to have a usable 0,0.

The rest of the details I (nor anyone else) don't know and cannot help you with, because it generally requires physical access to your printer to see how it moves and what works.

closed 2017-03-22 by lordofhyphens