slic3r/Slic3r-Manual #61: Layer number and height in gcode

rohrbrecher, 2018-11-12:
I am trying to use `before_layer_gcode` to insert comments into my gcode for easier M600/M125 insertion.

My current solution is working: `"; layer [layer_num], Z = [layer_z]mm"`

but I want the layer numbers to start at 1 instead of 0 (as in the preview window). 

Therefore I tried: `"; layer {[layer_num]+1}, Z = [layer_z]mm"`

Which gives me:

```
!!!!! Failed to process the custom G-code template before_layer_gcode
Parsing error at line 1. 
; layer {[layer_num]+1}, Z = [layer_z]mm
        ^
!!!!! End of an error report for the custom G-code template before_layer_gcode
```

PS: [layer_z] seems to have only one decimal, other than the Slic3r preview window :(

lordofhyphens, 2018-11-12:
Why is this a pull request against the manual?

lordofhyphens, 2018-11-12:
I think it needs to be `{{[layer_num]+1}}`

rohrbrecher, 2018-11-12:
Oh damn, I was browsing this part in the manual when I created the issue. I'll try the {{ }} right away. Thanks!

rohrbrecher, 2018-11-12:
Still no success:

```
!!!!! Failed to process the custom G-code template before_layer_gcode
Parsing error at line 1. 
; layer {{[layer_num]+1}}, Z = [layer_z]mm
        ^
!!!!! End of an error report for the custom G-code template before_layer_gcode
```

The manual just says: 

> Any expression enclosed in two curly brackets, but not starting with {if is evaluated as an arithmetic expression: {foo - bar}.

> All operators supported for conditional expressions are supported for value expressions.