slic3r/Slic3r-Manual #64: Document different Slic3r workflows

lordofhyphens, 2018-12-31:
Since tabs and dialogs are both options with Slic3r (and background processing can be toggled on/off), we should have a Manual page that documents a couple common workflows and why someone may want to use that specific way.

Here's a couple I can think of:
* Dialogs + Background Processing
    * Multiwindow editing (full profile editor)
    * live preview
* Tabs + background processing
    * Relies on the shortcut system
    * Live preview
* Background processing off
    * Less CPU intensive (less redrawing while options are thrashing about)

supermerill, 2019-01-02:
Background processing off:

* I load my part
* I go to print settings (my default one) and change many setting to suit my need for this print.
* i slice and check the preview & layers views, checking the bridges & overhangs.
* i  changes the settings until i'm satisfied.
* i export the gcode

huminlian, 2023-07-10:
This Manual doesn't support 3d printing slice settings for different [3d printer filament](https://www.sunlu.com/collections/3d-printer-filament).